webOS-ports/luneos-testing #2: Incorrect timezone is being chosen in FirstUse

Herrie82, 2022-01-25:
**Describe the bug**
The timezone chosen in FirstUse is incorrect, probably due to a change in the timezone data provided.

**To Reproduce**
In FirstUse select your country, when you come to the timezone page you'll notice it doesn't show or select the correct timezone.

**Expected behavior**
The correct timezone should be automatically selected based on country or available MCC/MNC when a SIM is present.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: All targets
 - OS: LuneOS
 - Version: 20220125

**Additional context**
Add any other context about the problem here.

Herrie82, 2022-01-30:
It seems that no offset is being populated for the countries anymore. Need to check luna-init to see if something needs updating in the timeZone files etc

Herrie82, 2024-01-02:
@Tofee It seems that "offsetFromUTC" ends up being 0 in all cases, causing the code in FirstUse not to work.

I guess the issue might be in here somehow?

https://github.com/webosose/luna-sysservice/blob/16eca0bf1b09bbbfeabbbb06c782b3167ea7979f/Src/TimePrefsHandler.cpp#L248-L251 which used to be:

https://github.com/openwebos/luna-sysservice/blob/c59b806a70bc1489a45f22454f07718bce51d350/Src/TimePrefsHandler.cpp#L1236-L1241

Tofee, 2024-01-02:
Mmh  both the new and the old code look good... It would need a pass in a debugger I think, to see more clearly what happens there.

Herrie82, 2024-01-02:
luna-send -n 1 luna://com.palm.systemservice/getPreferenceValues '{"key": "timeZone"}' also shows the incorrect value

Herrie82, 2024-01-02:
Ok seems issue might be in https://github.com/webosose/luna-init/blob/master/src/gen-ext-timezones.py

The generated ext-timezones.json doesn't contain the correct offsetFromUTC. Culprit is likely https://github.com/webosose/luna-init/blob/master/src/gen-ext-timezones.py#L121

Herrie82, 2024-01-02:
OK seems we need a new syntax in Python3: https://stackoverflow.com/questions/15940280/how-to-get-utc-time-in-python

I'll prepare a fix

Herrie82, 2024-01-02:
Seems I have a rebased and patched luna-init from OSE already in the works. Will test that first and add it to that PR

Herrie82, 2024-01-19:
Mostly solved now, just the selection of the correct timezone based on the country isn't working (anymore). Need to debug it a bit further:

https://github.com/webOS-ports/org.webosports.app.firstuse/blob/master/qml/TimeZonePage.qml#L169

Herrie82, 2026-09-06:
Solved by: https://github.com/webOS-ports/org.webosports.app.firstuse/pull/48

closed 2026-09-06 by Herrie82