webOS-ports/luneos-testing #5: Google C+DAV: Fix "Google hasn't verified this app"

Herrie82, 2022-01-25:
**Describe the bug**
Google upgraded it's security, we now need to verify our C+DAV Connector. Since the original developer is no longer actively involved, set it up under another account.

**To Reproduce**
Try to add Google C+DAV Account, you will get the warning screen.

**Expected behavior**
No warning screen.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: All
 - OS: LuneOS
 - Version: 20220125

**Additional context**
Add any other context about the problem here.

Herrie82, 2022-10-19:
@Garfonso Any chance you could share the settings you have the C+DAV in your Google Dev Console? Seeing we have to update the code anyway after Google's most recent change to OOB flow as perhttps://developers.google.com/identity/protocols/oauth2/resources/oob-migration 

It might be a good idea to take ownership of the credentials as well at our end using webos.ports@gmail.com so we can manage it from there.

Garfonso, 2022-10-20:
I'm willing to help, but I did not yet find out how to find that information on Googles Development stuff... if you can give me any further hints... 
Yes, it probably would be a good idea to take ownership with a webos ports account.

Herrie82, 2022-10-20:
@Garfonso Thanks as always :)

I'm also not sure yet, because I didn't set it up myself yet at our end:

https://console.cloud.google.com/home/dashboard should show you which API's you have I guess and their credentials and scopes?

Otherwise maybe in https://console.cloud.google.com/apis/dashboard or https://console.cloud.google.com/apis/credentials

Herrie82, 2022-10-20:
I got the invitation and accepted it, will take it from there :) Thanks for the help!