webOS-ports/luneos-testing #7: Web Browser won't re-launch

codepoet80, 2022-01-26:
**Describe the bug**
After minimal use the web browser stops working on Tenderloin

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh boot
2. User the web browser for awhile
3. Close the web browser (swipe card away)
4. Launch an enyo app (possibly one with a permissions error)
5. Close the app (swipe card away)
6. Try to launch the web browser

**Expected behavior**
Web browser can re-launch

**Smartphone (please complete the following information):**
 - Device: tenderloin
 - OS: LuneOS
 - Version: luneos-dev-image-tenderloin-20220125

**Additional context**
```
Jan 25 19:15:36 tenderloin LunaWebAppManager[1376]: DEBUG: 19:15:36.368: virtual luna::WebApplication::~WebApplication() luna::WebApplication(0x1c48458)
Jan 25 19:15:36 tenderloin LunaWebAppManager[1376]: LUNASERVICE ERROR 0: (null) ((null) @ (null):0)
Jan 25 19:15:36 tenderloin LunaWebAppManager[1376]: WARNING: 19:15:36.375: qrc:/qml/ApplicationContainer.qml:55: TypeError: Cannot read property 'internetConnectivityRequired' of null
Jan 25 19:15:36 tenderloin LunaWebAppManager[1376]: DEBUG: 19:15:36.384: virtual luna::WebApplicationWindow::~WebApplicationWindow() luna::WebApplicationWindow(0x23e6288)
Jan 25 19:15:36 tenderloin LunaWebAppManager[1376]: DEBUG: 19:15:36.453: Window destroyed
```

codepoet80, 2022-01-26:
Tried running: systemctl restart luna-webappmanager
Web browser still won't start

Tried running: systemctl restart luna-next  
Web browser still won't start

The only thing that resolves is a full reboot

Herrie82, 2022-01-26:
Well it seems to work OK here on emulator, so might be tenderloin specific

Herrie82, 2022-11-18:
We moved away from our own browser for now, using LG's with WAM and Chromium going forward. Closing this issue, might reopen it in the future again if needed with LG's browser.

closed 2022-11-18 by Herrie82