webOS-ports/luneos-testing #8: Apps can't launch other apps

codepoet80, 2022-01-26:
**Describe the bug**
Have been unable to get any of my apps to successfully interact with luna-send to launch another app. This includes explicit launches, like opening another app by its ID, and implicit launches, like invoking an email address or URL to be handled by a system app. I have ensured my app has the same permissions as system apps which CAN launch other apps.

**To Reproduce**
Steps to reproduce the behavior:
1. Install App Museum 2: http://appcatalog.webosarchive.com/latest.php
2. Modify the appinfo.json (/media/cryptofs/apps/usr/palm/applications/com.palm.app-museum2) to include permissions as shown here: https://github.com/codepoet80/webos-catalog-frontend/blob/main/appinfo.json 
3. Select an app, and scroll down to the support email or website and tap. 
4. See that nothing happens.

**Expected behavior**
When tapping a URL, the web browser should open. When tapping an email, the email client should open -- this works in the Contacts system app.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Tenderloin
 - OS: LuneOS
 - Version: 20220125

**Additional context**
App Museum call that does NOT work looks like:
`Jan 26 10:25:18 tenderloin LunaWebAppManager[1447]: [1447:1447:0126/102518.253859:INFO:CONSOLE(474)] "_webOS.exec(PalmSystem,LS2Call,[1,"luna://com.palm.applicationManager/open","{\"target\":\"mailto:test%40test.com\"}"])", source: userscript:webosAPI (474)`

System Contacts app call that DOES work looks like:
`Jan 26 10:28:07 tenderloin LunaWebAppManager[1447]: [1447:1447:0126/102807.333722:INFO:CONSOLE(474)] "_webOS.exec(PalmSystem,LS2Call,[4,"luna://com.palm.applicationManager/open","{\"target\":\"mailto:test%40test.com\"}"])", source: userscript:webosAPI (474)`

Error message is:
`<default-lib> LS_INVALID_HANDLE {"COND":"sh != NULL","FUNC":"_LSCallFromApplicationCommon","FILE":"callmap.c","LINE":1818} sh != NULL: failed`

Herrie82, 2022-01-26:
I've been able to track it down a bit so far, but not fully yet. It seems that somehow the file that should be generated by adding the requiredPermissions bit in the appinfo.json (/usr/share/luna-service2/client-permissions.d/com.palm.app-museum2.app.json) doesn't get created which causes the issues mainly. However I still didn't get it to work when manually adding it, so investigating it further now.

Herrie82, 2022-01-26:
I got it to work now, needed to add 3 files to get it to work.

/usr/share/luna-service2/client-permissions.d/com.palm.app-museum2.app.json
/usr/share/luna-service2/manifests.d/com.palm.app-museum2.manifest.json
/usr/share/luna-service2/roles.d/com.palm.app-museum2.app.json

In webOS OSE appinstalld2 is responsible for generating these permission files based on the requiredPermissions tag (https://github.com/webosose/appinstalld2/blob/master/src/installer/ServiceInstallerUtility.cpp). However we're still using luna-appmanager from Open webOS which doesn't have this functionality. So we'd need to migrate luna-appmanager in order for it to work. 

The apps we deploy currently are build using the image build process and we have a .bbclass which does the same basically, so therefore we don't have the issues with the other apps we deploy currently. 

I guess this one should get high on the priority list for things from OSE to migrate.

Interim solution might be to add the required bits to our luna-appmanager, but that might not be so easy and straightforward.

@Tofee any thoughts?

codepoet80, 2022-01-26:
Possible I could generate files during lune-install in sdk?

Herrie82, 2022-01-26:
> Possible I could generate files during lune-install in sdk?

That should be possible as a workaround for now, the code used during build is in:
meta-webos-ports/meta-luneos/classes/webos_app_generate_security_files.bbclass

Which I guess is python3 code.

Herrie82, 2022-01-26:
I might have something working for the emulator already, however time for bed here, so will test in the morning

Herrie82, 2022-01-27:
Seems it's working and installs it to the cryptofs location (at least on my emulator). Can you try the following?

1. Get this IPK for tenderloin:
https://www.herrie.org/appinstalld2_1.0.0-2+git0+3ea0bac9e7-r0_cortexa8t2hf-neon-halium.ipk

2. install it with "opkg install --force-reinstall --force-downgrade appinstalld2_1.0.0-2+git0+3ea0bac9e7-r0_cortexa8t2hf-neon-halium.ipk"

3. Restart TP just to be sure.

4. Then try to install the app-museum2 IPK with: luna-send -n 1 luna://com.webos.appInstallService/install '{"id": "com.palm.app-museum2", "ipkUrl": "/media/internal/com.palm.app-museum2_2.9.0_all.ipk", "subscribe": true}'

5. Do a luna-send -n 1 luna://com.palm.applicationManager/rescan '{}'
6. Afterwards you should have an icon in launcher and you should have the service files are generated and put in /var/luna-service2
7. The links to email and websites should now open

I've quickly tested this on the emulator and it seems to work there, but it still needs a bit more work probably for cleanup and proper integration, but it might be good to test already on your side as well.

codepoet80, 2022-01-27:
This command just hung forever on the command line:
luna-send -n 1 luna://com.webos.appInstallService/install '{"id": "com.palm.app-museum2", "ipkUrl": "/media/internal/com.palm.app-museum2_2.9.0_all.ipk", "subscribe": true}'

I tried a simpler install command, which did install the app:
luna-send -n 6 luna://com.palm.appinstaller/installNoVerify '{"subscribe":true, "target": "/media/internal/com.palm.app-museum2_2.9.0_all.ipk"}'

However, after running luna-send -n 1 luna://com.palm.applicationManager/rescan '{}' no files were generated in /var/luna-service2 and the app still cannot open links.

Herrie82, 2022-01-27:
You're not allowed to use the com.palm.appinstaller/installNoVerify anymore! That one doesn't generate the files.

Can you try "ls-monitor -i com.webos.appInstallService"
If that's stuck too try:
"systemctl restart appinstalld"
Then: 
"ls-monitor -i com.webos.appInstallService"

It should show you the available API calls then.

Afterwards try installing again.

I'm trying to get to the bottom of this issue at my side as well

Herrie82, 2022-01-27:
Somehow the systemd service file acts up, not really sure yet what's up with it, that's why you need to restart it manually. I'm looking into it. I changed quite some components (luna-appmanager to remove the bits that are now in ApplicationInstallerUtility from webOS OSE and appinstalld2), added those bits, which rely on ecryptfs which in turn relies on lttng, pmtrace and other bits. So what looks like a simple "change" of installer entails quite a bit in the end. I probably just have a small typo somewhere or some small permission file missing or something, question is where though... Hunting further

Herrie82, 2022-01-29:
OK this seems to be now close to ready for merge :)

https://github.com/webOS-ports/luneos-testing/releases/tag/20220129-LuneOS-Testing-Builds

Only issue for now is that you manually need to restart "appinstalld" still. Afterwards thing seem to be OK for me. 

App Museum installs, can open it and open the links.

Preware isn't updated for this yet, so when you install IPK via Preware it will still not work. The call needs to be updated there as well.

codepoet80, 2022-01-29:
OK, this works on the emulator image provided -- with some challenges:

+ You must restart the service before the first install (`systemctl restart appinstalld) after every boot, BUT I get permission denied if I try to do that via ssh. This means I can't script deployment.

+ Re-installing an app that's already installed requires a reboot before you can re-launch the app, and does NOT upgrade the app in place. This means you must uninstall an app before you re-install it.

+ Uninstalling an app has no effect until a reboot. I've tried calling `remove` on both the new and old appInstall services, and calling rescan afterward. The app remains in the launch until reboot. Only at that point can you re-install that app and get the new bits running on the device.

However, if I do all these steps I was able to launch the web browser, email client and Preware 2 from events in the App Museum!

codepoet80, 2022-01-29:
See these scripts that exercise all the steps in order. They work once per boot -- to run them a second time, you have to reboot:

https://github.com/codepoet80/webos-ports-sdk/blob/master/scripts/lune-run.sh
https://github.com/codepoet80/webos-ports-sdk/blob/master/scripts/lune-install.sh

Otherwise, things work. I suggest that maybe install/re-install lifecycle go into a new issue, and this issue can be closed -- since Apps _can_ launch other Apps now :-)

Herrie82, 2022-01-29:
Let me just tweak this a bit further. I think I know how to fix the remove issue, testing it now. The restart issue I'll check afterwards.

Herrie82, 2022-01-30:
> * e you can re-laun

The relaunch after reinstall not working was caused by a hardcoded appId that was incorrect (don't ask me where it came from, code from before my time I guess). This fixed it: https://github.com/webOS-ports/luna-appmanager/commit/d2b3bd6a625c168957482940a61d723980b85c5d 

I can open the app, re-install, which will close it, relaunch now and remove apps now here!

Only thing that still needs to happen after every step is the rescan still to update the launcher. I will try to see if there's another way somehow or if we can add it as a step maybe in the actions it does.

codepoet80, 2022-02-01:
I can install, launch, update/re-install and re-launch apps consistently! The Museum now works -- but Preware doesn't respond to the launch parameters. This is a new issue, I will work on.

Please use this version of the Museum for testing:
http://stacks.webosarchive.com/luneos/com.palm.app-museum2_2.9.0_beta_all.ipk

The SDK has been updated. I'll send that PR shortly, however, it would be good to have the `appinstalld` use its final name.

codepoet80, 2022-02-13:
Back to being unable to install apps on Tenderloin, fresh build as of 13-Feb-2022.

Output from: `opkg info appinstalld2`
```
Package: appinstalld2
Version: 1.0.0-35+git0+58faac369f-r0
Depends: applicationinstallerutility, ecryptfs-utils, libc6 (>= 2.34), libgcc1 (>= 11.2.0), libglib-2.0-0 (>= 2.68.4), libicuuc69 (>= 69.1), libpbnjson (>= 2.15.0-1+git0+6cd4815a81), librolegen, libstdc++6 (>= 11.2.0), luna-service2 (>= 3.21.2-1+git0+613f65e975), pmloglib (>= 3.3.0-2+git0+5907dc9dee)
Status: install ok installed
Architecture: cortexa8t2hf-neon-halium
Installed-Size: 798094
Installed-Time: 1644771095
```
Pushing an ipk to /tmp/ then running `/usr/bin/luna-send -n 6 luna://com.webos.appInstallService/install  '{"subscribe":true, "id": "$ipkname", "ipkUrl": "/tmp/com.palm.app-museum2_2.9.0_all.ipk"}'` gives output:
```
{"subscribed":true,"returnValue":true}
{"statusValue":24,"details":{"packageId":"","reason":"Failed to extract package","ipkUrl":"/tmp/com.palm.app-museum2_2.9.0_all.ipk","state":"install failed","verified":true,"installBasePath":"/media/cryptofs","client":"com.webos.lunasend-2031","errorCode":-5},"id":"com.palm.app-museum2"}
```

codepoet80, 2022-02-13:
I also tried manually installing the appinstalld2 version from above. opkg info output is:
```
sh-5.1# opkg info appinstalld2
Package: appinstalld2
Version: 1.0.0-2+git0+3ea0bac9e7-r0
Depends: libc6 (>= 2.34), libgcc1 (>= 11.2.0), libglib-2.0-0 (>= 2.68.4), libicuuc69 (>= 69.1), libpbnjson (>= 2.15.0-1+git0+6cd4815a81), libstdc++6 (>= 11.2.0), luna-service2 (>= 3.21.2-1+git0+613f65e975), pmloglib (>= 3.3.0-2+git0+5907dc9dee)
Status: install user installed
Architecture: cortexa8t2hf-neon-halium
Installed-Size: 732417
Installed-Time: 1644780169
```
No change in ability to install apps via `luna://com.webos.appInstallService/install `

jwise-cesmii, 2022-02-14:
New issue is not present on Hammerhead.
On Tenderloin, I tried:
- Pushing the IPK via SCP instead of ADB
- Installing the IPK manually from Preware
No change in error.

codepoet80, 2022-02-14:
New issue is not present on Hammerhead.
On Tenderloin, I tried:

-     Pushing the IPK via SCP instead of ADB
-     Installing the IPK manually from Preware'

No change in error.

jwise-cesmii, 2022-02-15:
Partially addressed by https://github.com/webOS-ports/appinstalld2/pull/6

Still need to do a `luna-send -n 1 luna://com.palm.applicationManager/rescan '{}'` before the app will launch.

codepoet80, 2022-02-16:
Only SDK-installed apps require rescan, Preware doesn't have this problem. Resolved by updating SDK, and closing.

closed 2022-02-16 by codepoet80