webOS-ports/luneos-testing #10: Preware 2 does not respond to lauch parameters

codepoet80, 2022-02-01:
**Describe the bug**
Preware 2 should handle launch parameters to facilitate other apps doing an install (like in Preware 1).

**To Reproduce**
Steps to reproduce the behavior:
1. Launch the App Museum II 2.9.0 beta for LuneOS: 
http://stacks.webosarchive.com/luneos/com.palm.app-museum2_2.9.0_beta_all.ipk
2. Choose an app to install and press the Download button
3. Observe the Preware Launches, but does not install the app requested in the launch parameters

**Expected behavior**
Preware should accept the launch parameters and confirm the user wants to install the specified file

**Launch Params**
params: { type: "install", file: "http://path.to/file" }

**Smartphone (please complete the following information):**
 - Device: Tenderloin
 - OS: LuneOS
 - Version: 20220130 

You can assign this to me, I can probably fix.

Herrie82, 2022-02-06:
I guess this is solved now? @Tofee did his magic on C++ side of things, seems we should be in a lot better state now for install & remove both via cmdline and Preware2. We'll push all the required changes tomorrow and provide a new build to test. 

The manual service restart stuff is solved too!

Herrie82, 2022-02-08:
The launchParams now work, so I guess we can close this one? The install is a separate issue we'd need to address properly.

closed 2022-02-08 by Tofee