webOS-ports/luneos-testing #11: Can't uninstall app from launcher

Tofee, 2022-02-06:
**Describe the bug**
When opening the launcher, and doing a long-press, a "cross" is available for user-installed apps that can be uninstalled.
But clicking on that cross doesn't work.
In the journal, I see:
LunaAppManager[1953]: cbRemove: name provided = [com.palm.app-museum2], getPackageInfoByPackageId FAIL, getPackageInfoByAppId FAIL - aborting remove

**To Reproduce**
Steps to reproduce the behavior:
1. Install a user app (through Preware2 for instance)
2. Open launcher
3. Long press on a user-installed app
4. Click on cross
4. See error in journal

**Expected behavior**
App should get uninstalled

**Screenshots**
N/A

**Smartphone (please complete the following information):**
 - Device: Pinephone
 - OS: LuneOS
 - Version: herrie/appinstalld2 branch of meta-webos-ports (feb 6th 2022)

**Additional context**
Add any other context about the problem here.

Herrie82, 2022-02-06:
@Tofee: We probably simply need to update this call: https://github.com/webOS-ports/luna-next-cardshell/blob/master/qml/LaunchBar/FullLauncher.qml#L669

Into something similar to this: luna://com.webos.appInstallService/remove '{ \"id\": \"$ipkname\", \"subscribe\": true }'" as per https://github.com/codepoet80/webos-ports-sdk/blob/master/scripts/lune-install.sh#L58

Tofee, 2022-02-06:
Yes, iirc it's what I wrote on the command line to uninstall the app manually, and it worked

Tofee, 2022-02-06:
Fixed with https://github.com/webOS-ports/luna-next-cardshell/pull/331

closed 2022-02-06 by Tofee