webOS-ports/luneos-testing #14: Apps can't save preferences: enyo.setCookie Does Not Work

codepoet80, 2022-02-16:
**Describe the bug**
App Museum II is an example of an enyo app that tries to use enyo.setCookie and enyo.getCookie to save and load preferences, but preferences are never set when running on LuneOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Install [App Museum II 2.9.0](http://stacks.webosarchive.com/luneos/com.palm.app-museum2_2.9.0_beta_all.ipk)
2. Launch the App, note that the About dialog appears. Dismiss the dialog by pressing its Close button.
3. Close the App.
4. Launch the App again, note that the About dialog re-appears.
5. Repeat the same steps on a webOS device or emulator. Note that after the first launch, the About dialog no longer appears at each launch.

**Expected behavior**
After the close button is pressed in the [About dialog](https://github.com/codepoet80/webos-catalog-frontend/blob/main/app/aboutPopup.js#L70), the App Museum [calls enyo.setCookie](https://github.com/codepoet80/webos-catalog-frontend/blob/main/init/banneret.js#L148) to save a preference indicating that first use is done. On subsequent launches, enyo.getCookie is called to read that preference. If set to true, the About dialog is not shown. On webOS, this works. On LuneOS the preference (apparently) doesn't get saved, so the dialog is shown every launch.

**Smartphone (please complete the following information):**
 - Device: Tenderloin, Hammerhead, and Emulator
 - OS: LuneOS
 - Version: Built 2022-02-15

**Additional context**
App Museum II [appinfo.json](https://github.com/codepoet80/webos-catalog-frontend/blob/main/appinfo.json#L22) _are_ set:
`"requiredPermissions": ["applications.internal", "applications", "activities.manage", "devices", "settings", "settings.read", "services", "database", "database.internal", "filecache", "system", "networking", "networking.internal"]`

Herrie82, 2022-02-16:
There are only a few clues in Open webOS's code for setCookie. I guess we might be missing something in QtWebEngine somehow. 

https://github.com/search?q=org%3Aopenwebos+setCookie&type=code

jwise-cesmii, 2022-02-16:
Of note, FeedSpider, which is a Cordova app, CAN save preferences on LuneOS -- but only because it uses localStorage and bypasses the enyo cookie methods:
https://github.com/OthelloVenturesLtd/FeedSpider2/blob/139544b9672331b83fa4ec2673de84b901a1d3d5/source/data/preferences.js#L241

codepoet80, 2022-02-16:
Of note, FeedSpider, which is a Cordova app, CAN save preferences on LuneOS -- but only because it uses localStorage and bypasses the enyo cookie methods:
https://github.com/OthelloVenturesLtd/FeedSpider2/blob/139544b9672331b83fa4ec2673de84b901a1d3d5/source/data/preferences.js#L241

Herrie82, 2022-11-18:
Issues are multiple, being Chrome (80+ mainly) no longer supporting cookies on "file://" because of spec violation and complete removal afterwards. I've worked around this by using LocalStorage in Enyo 1.0 when available. As per https://github.com/webOS-ports/enyo-1.0/commit/252012cf33432a45be6f9b939c3956fdf69e6e46

Suggest to close this one.

closed 2022-12-14 by Herrie82