webOS-ports/luneos-testing #15: Wi-Fi will not enable on HP Touchpad

Rupix6, 2022-02-23:
**Describe the bug**
None of the software toggles turn on Wi-Fi under the 20220125 Testing Build image (luneos-dev-image-tenderloin-20220125).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the pull-down menu next to the clock. "Wi-Fi" is in OFF status.
2. Click on 'Wi-Fi' to expand the sub-menu.
3. Tap the 'Turn on WiFi' link. The menu clears, but the Wi-Fi is not enabled.

Alternative method:
1. Tap the 'Launcher' icon at the far-right of the launch bar on the Home Screen. 
2. Tap the 'Prefs' tab to display the Preferences.
3. Tap the 'Wi-Fi' icon. Wi-Fi applet will launch to its App Menu screen, and the slider/toggle at the top-right corner is in OFF status.
4. Tap the top-right slider to the 'ON' position. The slider will change color to indicate it is enabled, but no networks display.
5. Tapping the '+ Join Network' option at the bottom-left of the App Menu window has no effect. (no error messages are issued)

**Expected behavior**
Enabling Wi-Fi will show a list of wireless networks which can be accessed.

**Screenshots**
No screen shot. No messages are issued to the display to capture.

**Smartphone (please complete the following information):**
 - Device: HP Touchpad 32 GB
 - OS: LuneOS
 - Version: 20220125

**Additional context**
- Airplane Mode is NOT enabled. Toggling its setting has no effect on Wi-Fi.
- The 'Turn on Bluetooth' toggle appears to operate normally from the clock/pull-down menu.
- LuneOS was installed following the instructions on https://forum.xda-developers.com/t/hp-touchpad-dual-boot-any-android-rom-and-luneos-the-new-webos.3999589/ .

Rupix6, 2022-03-09:
Found a bunch of messages in dmesg like this one:
[   184.251251] init: Command 'mkdir /data/misc/wifi/wpa_supplicant 0770 wifi wifi' action=post-fs-data (/init.rc:451) took 29ms and failed: lchown failed: Operation not permitted

The messages are similar when trying to run mkdir in the /data directory.

Herrie82, 2023-11-25:
This should be fixed in the new images out shortly

closed 2023-11-25 by Herrie82