webOS-ports/luneos-testing #18: Issues to fix before release

Herrie82, 2023-10-04:
After OSE rebase and with the (minor) fixes merged in https://github.com/webOS-ports/meta-webos-ports/pull/605
the following issues are still there that ideally we would address before the release:

- [ ] Simple Notifications don't work
- [ ] window.open or other mechanism is broken, launching multiple apps or 2nd app from 1st app gives black card without contents or the black screen can be brought up by minimizing the card (home/end buttons)
- [ ] geoclue not happy, but seems geoclue related, not specifically LuneOS. Error: Oct 04 13:11:00 qemux86-64 location-service[763]: Failed to start GeoClue2 client: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: 'location-getposition' disallowed, no agent for UID 0

Herrie82, 2023-10-04:
@Tofee ^

Tofee, 2023-10-05:
first issue is probably related to missing patch in SRC_URI in webruntime

Herrie82, 2023-10-31:
For GeoClue: Seems we need to build the agent and start it by default similar to: https://github.com/mosbasik/dotfiles/commit/d88a7c725c8fb9f64d7754c36df2b32a4b2bed75

Herrie82, 2024-01-06:
@tofee ^

Herrie82, 2024-01-15:
> * [ ]  Phone app: Call history is not populated.

`Jan 15 13:29:44 pinephonepro mojodb-luna[636]: [] [pmlog] DB8 MOJ_SERVICE_WARNING {"sender":"org.webosports.app.phone","method":"put","payload":{"objects":[{"_kind":"com.palm.phonecallgroup:1","callcount":1,"groupId":"org.webosports.app.phone pid.callgroup__PHONE_-4864402-56-13-+_15/1/2024_merge","recentcall_address":{"addr":"+31652044684","normalizedAddr":"-4864402-56-13-+","service":"com.palm.telephony"},"recentcall_type":"missed","timestamp":1705321784854,"timestamp_day":19737,"type":"merge"}]},"error":"schema validation failed for kind 'com.palm.phonecallgroup:1': invalid enum value for property 'type'","reqErr":-985}`

Also incoming SMS notification is not working with the following error (logical since we killed the org.webosports.notifications service I think?) Same issue would appear with update service that does a similar call:
```
2024-01-15T12:42:38.126138Z [1210.114921699] kern.err ports.service.messaging [] legacy-log  Error: palm://org.webosports.notifications/create {"ownerId":"org.webosports.service.messaging","launchId":"org.webosports.app.messaging","launchParams":{"threadId":"NDyfO+bU+ig"},"title":"+31652044684","body":"1 gemiste oproep(en) van +31652044684. \nLaatste oproep: 13:42 uur 15/01/2024.","iconUrl":"file:///usr/palm/applications/org.webosports.app.messaging/icon.png","soundClass":"notifications","soundFile":"","duration":-1,"doNotSuppress":false,"expireTimeout":5}: Service does not exist: org.webosports.notifications.     at Object.create (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:1309)     at /usr/palm/frameworks/foundations/version/1.0/node_module.js:1:35380     at Future.<anonymous> (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:10981)     at Timeout.d [as _onTimeout] (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:15961)     at listOnTimeout (node:internal/timers:559:17)     at processTim
2024-01-15T12:42:38.126800Z [1210.115574740] daemon.info ls-hubd[1409] Error: palm://org.webosports.notifications/create {"ownerId":"org.webosports.service.messaging","launchId":"org.webosports.app.messaging","launchParams":{"threadId":"NDyfO+bU+ig"},"title":"+31652044684","body":"1 gemiste oproep(en) van +31652044684. \nLaatste oproep: 13:42 uur 15/01/2024.","iconUrl":"file:///usr/palm/applications/org.webosports.app.messaging/icon.png","soundClass":"notifications","soundFile":"","duration":-1,"doNotSuppress":false,"expireTimeout":5}: Service does not exist: org.webosports.notifications.
2024-01-15T12:42:38.127160Z [1210.115931449] daemon.info ls-hubd[1409]     at Object.create (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:1309)
2024-01-15T12:42:38.127357Z [1210.116126574] daemon.info ls-hubd[1409]     at /usr/palm/frameworks/foundations/version/1.0/node_module.js:1:35380
2024-01-15T12:42:38.127514Z [1210.116282324] daemon.info ls-hubd[1409]     at Future.<anonymous> (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:10981)
2024-01-15T12:42:38.127663Z [1210.116431074] daemon.info ls-hubd[1409]     at Timeout.d [as _onTimeout] (/usr/palm/frameworks/foundations/version/1.0/node_module.js:1:15961)
2024-01-15T12:42:38.127802Z [1210.116569907] daemon.info ls-hubd[1409]     at listOnTimeout (node:internal/timers:559:17)
2024-01-15T12:42:38.127944Z [1210.116712532] daemon.info ls-hubd[1409]     at processTimers (node:internal/timers:502:7)
2024-01-15T12:42:38.128180Z [1210.116949657] daemon.info ls-hubd[1409] notification call had error: {"errorCode":-1,"response":{"returnValue":false,"errorCode":-1,"errorText":"Service does not exist: org.webosports.notifications."}}
2024-01-15T12:42:38.128455Z [1210.117226157] kern.err ports.service.messaging [] legacy-log  notification call had error: {"errorCode":-1,"response":{"returnValue":false,"errorCode":-1,"errorText":"Service does not exist: org.webosports.notifications."}}

```

Source: https://github.com/webOS-ports/org.webosports.messaging/blob/master/service/javascript/utils/MessageAssigner.js#L10

And for Update service: https://github.com/webOS-ports/org.webosports.update/blob/webOS-ports/webOS-OSE/service/javascript/assistants/checkUpdateAssistant.js#L157

Herrie82, 2024-01-17:
Audio not working, seeing this in logs:

```
Jan 17 13:16:51 pinephonepro com.webos.service.mediacontroller[1389]: [] [pmlog] mediacontroller MediaSessionManager {} setMediaMetaData mediaId : AC6C639E1B84336F170246F0AB5D4B8D
Jan 17 13:16:51 pinephonepro com.webos.service.mediacontroller[1389]: [] [pmlog] mediacontroller MediaControlService {} setMediaMetaData response : {"returnValue":true}
Jan 17 13:16:51 pinephonepro WebAppMgr[857]: [] [pmlog] wam.log WARNING {} W[857:857:on_observer_webos.cc(415)] "HandleMediaKeyEvent Session ID is not sent\n"
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: ALSA lib ../../alsa-plugins-1.2.6/pulse/pcm_pulse.c:758:(pulse_prepare) PulseAudio: Unable to create stream: No such entity
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: ALSA lib ../../../alsa-lib-1.2.6.1/src/pcm/pcm.c:8742:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: [1415:1415:0117/131651.090498:WARNING:alsa_util.cc(212)] PcmSetParams: default, Input/output error
Jan 17 13:16:51 pinephonepro WebAppMgr[1415]: [] [pmlog] wam.log WARNING {} W[1415:1415:io/alsa/alsa_util.cc(212)] "PcmSetParams: default, Input/output error\n"
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: ALSA lib ../../alsa-plugins-1.2.6/pulse/pcm_pulse.c:758:(pulse_prepare) PulseAudio: Unable to create stream: No such entity
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: ALSA lib ../../../alsa-lib-1.2.6.1/src/pcm/pcm.c:8742:(snd_pcm_set_params) Unable to set hw params for PLAYBACK: Input/output error
Jan 17 13:16:51 pinephonepro webapp-mgr.sh[1415]: [1415:1415:0117/131651.095577:WARNING:alsa_util.cc(212)] PcmSetParams: plug:default, Input/output error
Jan 17 13:16:51 pinephonepro WebAppMgr[1415]: [] [pmlog] wam.log WARNING {} W[1415:1415:io/alsa/alsa_util.cc(212)] "PcmSetParams: plug:default, Input/output error\n"

```

Tofee, 2024-01-17:
```Unable to create stream: No such entity``` is interesting, though I don't know what it might point to

closed 2026-09-05 by Herrie82