webOS-ports/luneos-testing #23: MTP no longer working

Herrie82, 2023-12-08:
**Describe the bug**
MTP mode is no longer showing on devices

**To Reproduce**
Connect via USB, MTP popup should show, however it doesn't 

**Expected behavior**
MTP popup to show and able to go into USB mode.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: mido, hammerhead, tenderloin, ppp
 - OS: LuneOS
 - Version: 20231208

**Additional context**
Probably got broken somewhere in recent updates

Herrie82, 2024-01-15:
Might be caused by https://github.com/webOS-ports/luna-next-cardshell/commit/73f964ba06af5d3930394b1b57e8c4fa791ed4d3 ?

Tofee, 2024-01-15:
Ah, maybe, I'll check who actually does the popup window

Herrie82, 2024-01-15:
@tofee Hmmz OK not sure then what's up.

Tofee, 2024-01-18:
I do get the popup on Touchpad. However, it's expected that it doesn't work on mainline devices, we don't have any support for these yet.

Then, https://github.com/webOS-ports/enyo-1.0/pull/23 should fix the popup behavior. However, note that mtp-server is still not started after clicking on the button.

Tofee, 2024-01-18:
It seems the mtp-server systemd unit is missing:
```
sh-5.1# systemctl start mtp-server
Failed to start mtp-server.service: Unit mtp-server.service not found.

```

Herrie82, 2024-01-19:
Fixed with https://github.com/webOS-ports/meta-webos-ports/pull/668

@codepoet80 ^

closed 2024-01-19 by Herrie82