webOS-ports/luneos-testing #27: LuneOS-Components: Unify display resolution settings

Herrie82, 2024-02-05:
**Describe the bug**
We currently define resolution in 2 different places which can be confusing.

https://github.com/webOS-ports/meta-webos-ports/blob/kirkstone/meta-luneos/recipes-webos-owo/luna-sysmgr-conf/luna-sysmgr-conf/raspberrypi3/luna-platform.conf#L21

https://github.com/webOS-ports/configd-data/blob/webOS-ports/webOS-OSE/configs/layers/base/raspberrypi3/com.webos.surfacemanager.json

The former is being used by LuneOS-Components: https://github.com/webOS-ports/luneos-components/blob/master/modules/LunaNext/Common/settingsadapter.cpp#L45-L54

Afterwards luna-next-cardshell uses this value as well. It would be better to get these values from configd instead.

Tofee, 2024-02-05:
I totally agree. However we must be careful: in settingsadapter it's a sync call, whereas with configd it will be async (LS2 call). So we might introduce a bug or two when doing that.

Herrie82, 2024-02-05:
> I totally agree. However we must be careful: in settingsadapter it's a sync call, whereas with configd it will be async (LS2 call). So we might introduce a bug or two when doing that.

I doubt the resolution will change much at runtime, so I don't see big issues there, but good to keep in mind.

Tofee, 2024-02-05:
I mean if there are some bugs, they'll mostly appear at startup time

Herrie82, 2024-02-05:
There seems to be a libconfigd which might be around just for this? 
Code: https://github.com/webosose/configd/blob/daab9ca0e43effc52c239d56387a6ca02e6dbb20/src/libconfigd/libconfigd.c

And some tests:
https://github.com/webosose/configd/blob/daab9ca0e43effc52c239d56387a6ca02e6dbb20/test_ref/tests/src/configd_client_libconfigd.c

Tofee, 2024-02-05:
Ah, well, why not; but let's try the simple LS2 way first

ericblade, 2024-02-22:
we might want to investigate if anyone's actually calling for that displayWidth/displayHeight .. or for that matter, an audit of anything referenced in that settings adapter -- I tweaked a few settings while trying to get RPi to boot in a different resolution, but this setting (as well as others like 'use virtual keyboard' that i noticed while digging in the file) seem to be completely unused.

Herrie82, 2026-09-05:
Solved with latest images dated late August 2026

closed 2026-09-05 by Herrie82