webOS-ports/luneos-testing #29: Multiple notifications and closure is not properly working

Herrie82, 2024-02-08:
Add multiple toasts (4 or so), delete the 3rd one and remaining ones aren't complete

Herrie82, 2024-02-09:
Example:

luna-send -n 1 -f -a com.webos.app.test luna://com.webos.notification/createToast '{"message":"hello world 1","displayId": 0,  "persistent":true}'
luna-send -n 1 -f -a com.webos.app.test luna://com.webos.notification/createToast '{"message":"hello world 2","displayId": 0,  "persistent":true}'
luna-send -n 1 -f -a com.webos.app.test luna://com.webos.notification/createToast '{"message":"hello world 3","displayId": 0,  "persistent":true}'
luna-send -n 1 -f -a com.webos.app.test luna://com.webos.notification/createToast '{"message":"hello world 4","displayId": 0,  "persistent":true}'

Afterwards remove "hello world 3" and a random notification gets removed from the list it seems

Tofee, 2024-02-09:
Thanks, with this scenario it should be easy to find the mistake

Herrie82, 2026-09-22:
Should be solved with https://github.com/webOS-ports/notificationmgr/pull/3

Herrie82, 2026-11-11:
Solved

closed 2026-11-11 by Herrie82