webOS-ports/luneos-testing #31: Hide VKB when physical keyboard is available (qemux86. PP(P), PT2)

Herrie82, 2024-02-14:
Currently VKB shows even when HW kbd is available. This takes unnecessary screen real estate.

Make sure to auto-hide it. See https://github.com/webosose/ime-manager/blob/master/maliit-plugin-global/plugin/inputmethod.cpp for some example implementation.

Might need to think of a way to bring up VKB somehow.

Tofee, 2024-02-14:
Note: this behavior is caused by our plugin being registered for both OnScreen and Hardware states of Maliit. Using imemanager for Hardware state caused conflicts in the management of the OnScreen state, which is also claimed by the "global" plugin of imemanager.

Herrie82, 2024-02-21:
I guess the right way forward would be to migrate webos-keyboard to ime-manager

Herrie82, 2026-11-11:
Solved recently

closed 2026-11-11 by Herrie82