webOS-ports/luneos-testing #34: Preware feeds not loading

Herrie82, 2024-02-20:
Preware feeds are not being loaded. Not sure what's the cause, could be FirstUse not tagging them correctly or some issue parsing the feeds (due to -webos suffix by WAM/Chromium?)

Herrie82, 2024-02-22:
@tofee Did some debugging here, seems that we get a wrong version back from deviceinfo. It returns "00.00.00" now. There is a workaround available in Preware code that checks for "0.0.0", updating this to "00.00.00" works around the issue. It's more interesting though to see why we don't get anything populated properly in DeviceInfo:

I'm seeing this when logging the PalmSystem.DeviceInfo:
{"modelName":"webOS.Open","platformVersion":"00.00.00","platformVersionDot":0,"platformVersionMajor":0,"platformVersionMinor":0,"screenHeight":0,"screenWidth":0}

Which seems to come from WAM: https://github.com/webosose/wam/blob/ff3b892d086cb759b2cc6160f9a85f221317deab/src/webos/device_info_impl.cc

It seems to parse some file, but I'm not sure which one....

https://github.com/webOS-ports/preware/blob/fc3c5ca47638c66a22a1d4c280275bd0dc5879d5/source/model/packagesModel.js#L294

ericblade, 2024-02-23:
wam doesn't actually retrieve any values to set in there from anywhere, other than localcountry, smartservicecountry, and screenrotation ... at least, not in any code that i see.  is there a place it should get that data from?

Herrie82, 2024-04-10:
In recent builds had issues with feeds not showing up in FirstUse/Preware. Not sure of the root cause (initscripts and/or scarthgap), but fixed with this: https://github.com/webOS-ports/meta-webos-ports/pull/712

ericblade, 2024-04-11:
even if that change not related to the version fixes preware, the version information still should be fixed, as well as any other related calls in there that are broken...

Herrie82, 2024-04-11:
> even if that change not related to the version fixes preware, the version information still should be fixed, as well as any other related calls in there that are broken...

Well @Tofee fixed that in WAM with this :) https://github.com/webOS-ports/meta-webos-ports/pull/687/files