webOS-ports/luneos-testing #37: tenderloin: Get mainline kernel working

Herrie82, 2024-04-17:
Seeing 3.4 kernels are really EOL now by the looks of it, we need to get our 3.4 targets (mako, hammerhead, tenderloin) working with mainline kernel. 

There were various attempts done in the past:
https://github.com/mboudr35/hp-touchpad-linux
https://github.com/lopsided98/linux/tree/hp-tenderloin
https://git.kernel.org/pub/scm/linux/kernel/git/linusw/linux-integrator.git/log/?h=apq8060-dragonboard-graphics
https://github.com/shr-distribution/linux/tree/tenderloin/5.19/mainline

Steps: 
1. Submit missing bits to upstream kernel for consideration:

- mmcc bindings: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240417-msm8660-mmcc-v1-1-efc9e455268c@herrie.org/

- dt-bindings: phy: qcom,usb-hs-phy: Add compatible: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240417065454.3599824-1-github.com@herrie.org/

- dt-bindings: mfd: qcom: pm8xxx: Add pm8901 compatible: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240415123038.1526386-1-github.com@herrie.org/

- mfd: pm8xxx: add support for PM8901: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240415123941.1527938-1-github.com@herrie.org/

- dt-bindings: pinctrl: qcom,pmic-mpp: add support for PM8901: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240417073532.3718510-1-github.com@herrie.org/

- pinctrl: qcom-ssbi: add support for PM8901: https://patchwork.kernel.org/project/linux-arm-msm/patch/20240415141814.1983384-1-github.com@herrie.org/
 
2. Update the device tree and submit upstream

Herrie82, 2024-11-11:
Various schematics (mainly Opal and Topaz 3G, so no Topaz WiFi only)
[Schematics-Opal.6050A2430401.rX01.pdf](https://github.com/user-attachments/files/17699371/Schematics-Opal.6050A2430401.rX01.pdf)
[Schematics-Opal_6050A2430401-mb_dvt_0704.pdf](https://github.com/user-attachments/files/17699372/Schematics-Opal_6050A2430401-mb_dvt_0704.pdf)
[Schematics-Topaz3G-6050A2428701-MB-X01.20101012.pdf](https://github.com/user-attachments/files/17699374/Schematics-Topaz3G-6050A2428701-MB-X01.20101012.pdf)
[Tech-Specs-hptouchpad-120110090301-phpapp02 (1).pdf](https://github.com/user-attachments/files/17699375/Tech-Specs-hptouchpad-120110090301-phpapp02.1.pdf)

Herrie82, 2026-01-20:
@J0SH1X I came across your kernel repos for the slightly older generation SoC from qcom, thought above work could be of interest to you too!

Herrie82, 2026-01-20:
> also what do you mean by mainline kernel like upstreaming old devices or updating the ancient kernel.

6.18 upstream kernel with all patches for HP Tenderloin (HP TouchPad) added. It's booting, display seems to work and modetest/kmscube now work (fps leaves to be desired). We need to test everything else still more or less, but I was able to create this mostly with Claude code, 2.6 kernel source code and a device running legacy HP/Palm webOS with 2.6 kernel! 

You could do the same for your HD2 platform :) Some things might be very similar, hence I pinged you.

BTW: LG took over webOS from HP and now uses it for TV's ;)

Herrie82, 2026-01-20:
@robclark This one refuses to die just :) 6.18 kernel up and running. Modetest looks good without artifacts! Will send some patches I guess later for the kernel driver :) Just on kmscube we get about 0.5 to 1.0 fps. I would expect this should run better? You have any thoughts as to what is an expected fps in kmscube for this? Any thoughts why it might be so slow? We have CPU and GPU scaling enabled, it seems to properly scale up to 270Mhz.

BTW this is using CMA, IOMMU still having some issues we're looking into.

robclark, 2026-01-20:
> [@robclark](https://github.com/robclark) This one refuses to die just :) 6.18 kernel up and running. Modetest looks good without artifacts! Will send some patches I guess later for the kernel driver [...]

I would expect a lot more than that, even on a slow device

What output do you get from kmscube?  And dmesg?

mboudr35, 2026-01-30:
Hi! Thanks for the poke. Unfortunately my attention is taken elsewhere but I'm very happy to see the progress that's been made! Best of luck!