webOS-ports/luneos-testing #38: Mako: Bootlooping with weird graphical artifacting

nothingneko, 2024-09-16:
**Describe the bug**
Instead of booting to LuneOS, my Nexus 4 bootloops and has some weird graphical artifacting
**To Reproduce**
Steps to reproduce the behavior:
1. Started on stock firmware 5.1.1
2. Flashed a copy of TWRP, and used TWRP to flash Clockwork (the IMGs of Clockwork no longer are available) 
3. Flashed Cyanogen mod from Clockwork, rebooted and set it up
4. Enable dev mode
5. Reboot to recovery, factory reset and sideload LuneOS build 20230514
6. Boot to OS and see strange artifacting before the phone shuts down

**Expected behavior**
The device boots to LuneOS

**Screenshots**

these are in order, i took a video and pulled these out

[image]

[image]

[image]

[image]

[image]

[image]

**Smartphone (please complete the following information):**
 - Device: Mako
 - OS: LuneOS
 - Version: 20230514

**Additional context**
Add any other context about the problem here.

Herrie82, 2024-09-16:
Hi, try the latest release image instead please: https://github.com/webOS-ports/luneos-releases/releases/

nothingneko, 2024-09-16:
Will try that tonight!

Herrie82, 2026-11-13:
Should be working OK with latest 3.4 based kernel image.

closed 2026-11-13 by Herrie82