webOS-ports/luneos-testing #39: Tenderloin screen is very dim

markfmasters, 2025-01-08:
**Describe the bug**
The screen is very dim, even if  at maximum brightness

**To Reproduce**
Steps to reproduce the behavior:
It is like this from boot.

**Expected behavior**
Previous releases have been significantly brighter.  It would be good if there was more dynamic range to the brightness

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Tenderloin
 - OS: LuneOS
 - Version: the most recent version

**Additional context**
I am new to LuneOS but have a number of HP touchpads and the most recent release is really good.

Herrie82, 2025-01-08:
Hi Mark,

Did you try the official release (https://github.com/webOS-ports/luneos-releases/releases/tag/Eiskaffee) or test?

markfmasters, 2025-01-08:
Hi Herrie,

I am using the official release.

Mark

Herrie82, 2025-01-08:
Hi Mark,

The problem with Tenderloin is that it is using a very old (Franken) kernel version 3.4 (It originates from a HTC device not even the TouchPad itself, since HP only ever released kernel 2.6). 

We're currently doing many underlying OS updates which will mean there will be no longer support for devices with kernel version 3.4. 

We're looking into getting Tenderloin working with a mainline kernel (6.10+), however we are currently having some issues with the exotic Cypress touchscreen that is being used. We're close to get it somehow working, but not there yet. 

Fixing the current build based on 3.4 kernel is not really an option at the moment.

There are some hacks that could work to improve the situation. I'll dig them up and post them here.

markfmasters, 2025-01-08:
Herrie,

First, thank you for the hacks.  I appreciate that.

Second, I completely understand that the 3.4 kernel cannot be used any longer.  I had read about that issue in the notes.

Third, I would be happy to assist with the touchscreen if you can direct me how to get started.  I have worked with cypress circuitry in the past, so I may have some insights?

I would play around with building my images, but with webos-ports.org not functioning, I cannot get started.

The 20240214 release is really good overall.  I am very impressed.  It is very functional.

Last

Mark

Tofee, 2025-01-08:
Hi @markfmasters 

I've been tinkering with the tenderloin's touchscreen for a few weeks, and some help is always welcome :)
The cypress chipset here is a CY8CTMA395, probably a "Gen3" cypress chip. It's can be accessed both via I2C and via UART from the main msm8660 chip.

In legacy webOS, the initialization is done in the following (weird) way:
 - both i2c and UART are activated on the board for these pins, in a "shared" mode
 - the regulators are brought up
 - i2c is used first, but only to setup some of the values on the bus
 - the actual data is received via UART at 3Mhz

I've tried to follow the same procedure on mainline: I receive some data on UART, but it's mainly zeroes, so it can't be actually decoded.

I've tried reusing the generic Gen3 cypress driver of the 6.10 kernel, but it fails to initialize the device.

Watching the i2c bus with i2cdump, I can't see any indication that the touch data are available there. Almost all the values are constant, and there's a counter increasing (maybe a timestamp).

I'm now trying to debug side-by-side the android kernel (3.4) and the mainline kernel (6.10), to analyze what we send via the msm_serial calls. It's still WIP.

Of course we don't have any documentation whatsoever, which doesn't really help... However we do have the board design, available in pdf here: https://github.com/webOS-ports/luneos-testing/issues/37#issuecomment-2467519818

Herrie82, 2026-09-29:
Fixed in recent builds that contain proper suspend and sleep

closed 2026-09-29 by Herrie82